adambujak/SignBuddy #27: Organize applications directory

stephen99scott, 2021-11-13:
The applications/SignBuddy directory has quite a few files now, should we consider restructuring the directory and add subdirectories like `src` and `inc`?
* `applications/SignBuddy/src/adc.c` 
* `applications/SignBuddy/inc/adc.h`
* etc...

We could leave main.c in `applications/SignBuddy/main.c`

adambujak, 2021-11-13:
i don't like leaving main separate from the rest of the source files.

I used to like the project structures with  `src`, `include`, and `build` directories, but I've grown to prefer everything just being in the main application directory.

My reasoning is that in general I find it quicker to add new modules, find modules, and build when everything is simply in the top level. In the past I've found it very annoying to be in the src folder and have to remember to type `vim ../same_source.c` or be in the top level and have to specify whether its in the src or include directory.

adambujak, 2021-11-13:
I'm open to using the other project structure, but I would prefer to just leave it the way it is right now

stephen99scott, 2021-11-13:
> very annoying to be in the src folder and have to remember to type `vim ../include/same_module.h` or be in the top level and have to specify whether its in the src or include directory.

Yeah this is understandable. I'm fine with leaving it as is, just wanted to see what your thoughts were on it.

closed 2021-11-13 by stephen99scott